powtac/Autarc_LAN_User_Stats #56: New URLs for web service

powtac, 2015-03-21:
#### Required APIs on web service
(See https://github.com/powtac/Autarc_LAN_User_Stats/issues/10#issuecomment-55624432)
1. Receive Requests
2. Return vendor/manufacture for a given MAC address / Return name for an device by MAC address 
3. List all scanners
4. List the stats for a single scanner
5. Set the name for a device

### `POST  [network_name]/ping_result/add`
```JSON
{
	"network_name": "Fischergasse",
	"online": [
		{"ip": "10.0.1.12", "t": 0, "mac": "111:111:111:111:111"}
	],
	"offline": [
		{"ip": "10.0.1.8",  "t": 13},
		{"ip": "10.0.1.9",  "t": 7},
		{"ip": "10.0.1.10", "t": 4},
		{"ip": "10.0.1.11", "t": 2}
	]
}
```
`t`  = time ago in seconds

### `GET  [network_name]/device/[mac]/info`
```JSON
{
	"network_name": "Fischergasse",
	"mac": "111:111:111:111:111",
	"mac_formated": "111:111:111:111:111",
	"vendor": "Apple",
	"name_set": TRUE,
	"name": "Simon iPhone",
	"last_seen_seconds": 233534534,
	"last_seen_timestamp": 80346456456,
	"last_seen_datetime": "12.12.2015 18:35"
}
``` 
`last_seen_seconds`  = time ago in seconds
`last_seen_timestamp`  = unix timestamp
Maybe a `history` object will be added later

### `GET  networks/list`
```JSON
{
	"networks": ["Fischergasse", "TIM ARM"]
}
```
*JSON is always an object*

### `GET  [network_name]/stats[/range]`
```JSON
{
	"network_name": "Fischergasse",
	"range": "12 days",
	"data": [
		{"t": 5466, "d": "23.12.2015 03:20:12", "ip": "10.0.5.13", "status": "off",},
		{"t": 5466, "d": "23.12.2015 03:20:12", "ip": "10.0.5.4", "status": "on", "mac": "111:111:..."}, 
		{"t": 5478, "d": "23.12.2015 03:20:17", "ip": "10.0.5.14", "status": "off"},
		{"t": 5489, "d": "23.12.2015 03:20:20", "ip": "10.0.5.15", "status": "off"},
		{"t": 5490, "d": "23.12.2015 03:20:22", "ip": "10.0.5.16", "status": "off"},
		{"t": 5456, "d": "23.12.2015 03:20:24", "ip": "10.0.5.17", "status": "off"},
		{"t": 5445, "d": "23.12.2015 03:20:25", "ip": "10.0.5.18", "status": "off"},
		{"t": 5465, "d": "23.12.2015 03:20:26", "ip": "10.0.5.19", "status": "off"},
		{"t": 5464, "d": "23.12.2015 03:20:29", "ip": "10.0.5.20", "status": "off"},
		{"t": 5445, "d": "23.12.2015 03:20:32", "ip": "10.0.5.21", "status": "off"},
		{"t": 5425, "d": "23.12.2015 03:20:36", "ip": "10.0.5.22", "status": "off"},
		{"t": 5489, "d": "23.12.2015 03:20:38", "ip": "10.0.5.23", "status": "off"},
		{"t": 5456, "d": "23.12.2015 03:20:39", "ip": "10.0.5.24", "status": "off"},
		{"t": 5463, "d": "23.12.2015 03:20:40", "ip": "10.0.5.25", "status": "off"},
		{"t": 5495, "d": "23.12.2015 03:20:44", "ip": "10.0.5.26", "status": "off"},
		{"t": 5456, "d": "23.12.2015 03:20:45", "ip": "10.0.5.27", "status": "off"},
		{"t": 5466, "d": "23.12.2015 03:20:46", "ip": "10.0.5.28", "status": "off"},
		{"t": 5489, "d": "23.12.2015 03:20:47", "ip": "10.0.5.29", "status": "off"},
	]
}
```
`range` is http://php.net/manual/en/function.strtotime.php (default: "48 h")
Data points on the same time will become two data points

### `POST  [network_name]/device/[mac]/name`
```JSON
{
	"network_name": "Fischergasse",
	"mac": "111:111:111:111:111",
	"name": "Simon iPhone"
}
```

powtac, 2015-03-24:
All APIs are now implemented on http://DOMAIN.COM/autarc_lan_user_stats/03/?/debug/phpinfo for example

powtac, 2015-06-05:
This shows how it works: `POST` with JSON in HTTP Body and additional Header `Content-Type	
application/json; charset=UTF-8`

[image]

[image]

PcTim, 2015-06-05:
Should we also send additional information about the devices to the server?
e.g. DNS server, Gateway, Arduino board

powtac, 2015-06-05:
Good idea! But I don't want to store sensitive data like Gateway/DNS Server on the web But Arduino Version... is a good idea.

PcTim, 2015-06-05:
I didn't meant Arduino Version, i meant if the scanned IP is the one of the Arduino board...
Should we filter gateway, dns server and the arduino IP address and don't send any stat about them to the server?

powtac, 2015-06-05:
Ah! Sure, but much later. It is not important right now and also somehow useful to know that DHCP, ... is there :-)

commit 9c325b37 (2015-09-26): Link device specific statistic at Arduino webpage

all new URLs are included/linked now.
closes #56
closed 2015-09-26 by PcTim