powtac/Autarc_LAN_User_Stats #58: Server doesn't accept data

PcTim, 2015-06-05:
The server doesn't accept "Content-Type: application/json;" from the arduino.
Also we may have problems with the ? in the servers-path.

The server returns only a 403 Forbidden error.
I think the servers mod security doesn't allow POST data within a request, but I'm not sure. Powtac, could you post the servers access data, please?

PcTim, 2015-06-05:
I've tried in different combinations:
"Content-Type: application/json;"
"Content-Type: application/x-www-form-urlencoded;"
"Content-Type: application/x-www-form-urlencoded; charset=UTF-8;"
"X-Requested-With: XMLHttpRequest"

...but all without having success

PcTim, 2015-06-05:
Also I've tried it with dummy data and calculated Content-Length the whole time. It only succeed in getting no 408 Request-Time-Out error.

powtac, 2015-06-05:
I changed the server, please try again.

PcTim, 2015-06-05:
Seems good! I get a response! Now I've to look why the data won't be stored, but shouldn't be that problem...

closed 2015-06-05 by PcTim

powtac, 2015-06-05:
When you printout the debug info of the response of the server, can you print out the URL as well? I do test it now as well with my hardware...

commit 35a674ce (2015-06-05): bugfix JSON data

fix #58