powtac/Autarc_LAN_User_Stats #62: Autarc_LAN_User_Stats stopped working after +-12 h

powtac, 2015-06-07:
While doing the first real world test the setup stopped working after about 12-14 hours. 

@PcTim suggests it might be connected to the DHCP renewal. I think this could be solved by calling `Ethernet.maintain();` http://www.arduino.cc/en/Reference/EthernetMaintain every some hours. Or maybe when starting a new IP scan loop.

Another idea was that it stopped working because of some memory overflow.

powtac, 2015-06-07:
I think I will do a long term test with opened Arduino IDE Serial Monitor attached.

PcTim, 2015-06-07:
If I'm right, <code>Ethernet.maintain();</code> is called. I'll have a look at this problem. Doing a long term test with the serial monitor is a good idea. Please post the last few lines. There should be a note that the Ethernet connection failed and the board tried to reconnect.

PcTim, 2015-06-07:
<code>Ethernet.maintain();</code> is called. Also it should be reconnect by itself. Please check the Serial print

powtac, 2015-06-08:
Currently the tracker is functioning about 24h in a row!

powtac, 2015-06-12:
I think I found the problem: The Apache server crashed tonight and around the same time the tracker completely stooped to work. So this seem to crash the tracker. Can could that be solved by an timeout on the client that stops waiting after let's say 1-2 minutes when the server does not respond?

PcTim, 2015-06-13:
Great!
Hmm... A crash shouldn't cause this problem.
When a connection to the server fails, the board tries it x times (can be configured in setup). If it also fails, the board will check its own network connection by pinging the gateway. If the gateway is available it will retry it again to connect to the server and so on...
If the gateway is offline, the board tries to start a new connection.

PcTim, 2015-06-13:
Oh! I'll see! It's a recursive function! Probably we have a buffer overflow after to many fails!
I'll check how to solve it!

PcTim, 2015-06-13:
Can you check it again please?

powtac, 2015-06-13:
I will check it soon. Not sure when. Maybe tomorrow. Thanks for fixing it!

PcTim, 2015-08-11:
Have you checked yet?
I'll have a try at #65 so it would be easier to check possible issues.

powtac, 2015-08-11:
I think it worked. The tracker is still running, but I'm currently not sure on which firmware...

commit 1d1622a6 (2015-06-13): deleted recursive call of send_info_to_server

maybe fixes #62