powtac/Autarc_LAN_User_Stats #63: ICMPPing.cpp TTL problem

powtac, 2015-06-07:
See https://github.com/BlakeFoster/Arduino-Ping/issues/5

Is https://github.com/BlakeFoster/Arduino-Ping/issues/5#issuecomment-86068702 something that you can do?

PcTim, 2016-04-16:
I think we're having no problems with the TTL. We don't analyze this value. As I can see all my devices were recognized well. The main problem we have is the duration of one ping command.
I've checked their new async function (https://github.com/BlakeFoster/Arduino-Ping/commit/68201bb684871cddee5de9fcb0354fe6015b485b)

PcTim, 2016-04-16:
But I'll check how we can decrease this ping-times.
In their example(https://github.com/BlakeFoster/Arduino-Ping/blob/master/icmp_ping/examples/Ping/Ping.ino) they request a ping every 500ms. I'll check how they have solved it.

PcTim, 2016-04-16:
I have checked their example - if the pinged device is answering we truly have such short times, but if the device isn't responding we also need about 10 seconds. This is much too long! We have to decrease this time!
references #53 #16

PcTim, 2016-04-16:
The long time has three causes:
1. If we ping an offline device the Ethernet-Controller gets an error because noone answers the request. So the Ethernet-Controller replies after about one seconds an sending-timeout error.
This sending-timeout time is a different one than the ping-timeout time. The ping-timeout specifies the time in which an answer from the device and an ping answer has to been recieved.
The main problem is that we can only set the ping-timeout time. The sending-timeout value is specified decoded in the w5100 library which is operated by the arduino library... So I think we unfortunately have no chance to set down this time.
2. According to the number of ping-requests which the user can set up we try as default four times to ping an device. By setting this value to 1 during the initial configuration we can reduce the scanning time to a quarter.
3. We have a very long Server-Listen-Delay (about 4 seconds) after each ping request. This takes much of our time! We need such a long delay that the arduino server is reachable from web browsers but I think we could try to replace this loop and useless waiting with an Interrupt-event.
#53  #16 

So the TTL is currently not a problem to us. I'll open a new Issue for the third point.

closed 2016-04-16 by PcTim