powtac/Autarc_LAN_User_Stats #64: Same Mac address used by two different IP addresses

powtac, 2015-06-07:
This is probably a bug in the code, the Mac Address is not filled correctly in the second case...
[image]

PcTim, 2015-06-07:
Is it possible, that the device has changed its IP address during the scan. (Within 6 minutes it would be possible) also the board has had problems with the server connection as we can see in the time difference within the online device.

PcTim, 2015-06-07:
Oh! Hm. Strange: IP .1.6 was scanned 19:30:09; IP .1.32 was scanned 19:33:15 - 186 seconds = 19:30:09

PcTim, 2015-06-07:
Maybe a buffer problem...

PcTim, 2015-06-07:
Have you had this problem twice?
Do you know if there truly was a device at both IPs? A had a suggestion, that there was a mistake in the ping libary. #63

powtac, 2015-06-08:
> Have you had this problem twice?

Yes, the problem occurs several times in the log files!

PcTim, 2015-06-13:
Already after the previous commit?

powtac, 2015-06-13:
I will check it again with the latest version of code...

PcTim, 2015-09-26:
Should be fixed since the b5097cc868c27b66dc830c5c379c6807074c4332 commit

closed 2015-09-26 by PcTim